ros-controls/ros2_control_ci #27: coverage build: Commit creating failed

christophfroehlich, 2024-03-02:
```
==> Running command '/home/runner/work/_actions/codecov/codecov-action/v4/dist/codecov create-commit'
/home/runner/work/_actions/codecov/codecov-action/v4/dist/codecov create-commit --git-service github -C 2c655e53579d5dbd22ccd56be00cfb980c8738ee -Z
info - 2024-03-01 10:01:55,492 -- ci service found: github-actions
info - 2024-03-01 10:01:55,658 -- Process Commit creating complete
error - 2024-03-01 10:01:55,658 -- Commit creating failed: {"detail":"You do not have permission to perform this action."}
Error: Codecov: Failed to properly create commit: The process '/home/runner/work/_actions/codecov/codecov-action/v4/dist/codecov' failed with exit code 1
```
https://github.com/ros-controls/ros2_controllers/actions/runs/8109385113/job/22164449480?pr=1066

@fmauch any ideas what that means?

fmauch, 2024-03-02:
Phew, that might be due to the mergify bot?

The codecov-action docs mention [special treatment for dependabot](https://github.com/codecov/codecov-action?tab=readme-ov-file#dependabot), that could also be necessary for mergify.

christophfroehlich, 2024-03-02:
the question is, do we need that for the pull_request event at all? let's see if it works after the first mergify PR got merged.

christophfroehlich, 2024-03-02:
OK, the latest merge to the master failed with the same error. https://github.com/ros-controls/ros2_controllers/actions/runs/8125063697/job/22207239926
I updated the token now, we'll see

fmauch, 2024-03-02:
https://github.com/ros-controls/ros2_controllers/actions/runs/8125723976/job/22208663170 seems to be working.

christophfroehlich, 2024-03-03:
you are right. but what changed that ros2_controller has >70% coverage? :see_no_evil: