ros-controls/ros2_control_ci #33: ICI workflow: Cache is shared between different jobs

christophfroehlich, 2024-03-14:
> if the jobs are unique for different branches, then this would solve the case from above?

Not necessarily. Caches from PRs and scheduled builds would still get mixed up. That might be fine in most cases, but might as well lead to strange errors as the one mentioned above.

I guess, the cleanest would be:

- Use a proper job identifier (e.g. `github.workflow`)
- Use three possbile restore keys: 
  -  `...${{ github.base_ref }}-${{ github.ref }}`
  - `...${{ github.base_ref }}`
  - `...${{ github.ref }}`

This way on a PR it would search for a cache e.g. `...master-my_feature` and if it doesn't find that during the first build use `...master`, while for a scheduled build it would use `...master`, since `...master-` and `...` don't exist.

That could work, right?

Changing things to actual actions rather than reusable workflow could be a separate discussion, I think.

_Originally posted by @fmauch in https://github.com/ros-controls/ros2_control_ci/issues/30#issuecomment-1997049189_

christophfroehlich, 2024-04-16:
Summarizing [tests](https://github.com/ros-controls/ros2_control_ci/actions/runs/8708071137/job/23884577080?pr=49)

```yml
name: Reusable test contexts

on:
  workflow_call:

jobs:
  reusable-test:
    runs-on: ubuntu-latest
    steps:
      - run: |
          echo "Reusable Github.job ${{ github.job }}"
          echo "Reusable Github.workflow ${{ github.workflow }}"
```
```yml
name: Test contexts

on:
  workflow_dispatch:

jobs:
  test:
    runs-on: ubuntu-latest
    steps:
      - run: |
          echo "Github.job ${{ github.job }}"
          echo "Github.workflow ${{ github.workflow }}"
  call_reusable_test:
    uses: ./.github/workflows/reusable_test_contexts.yml
```

gives 
```console
| Github.job test
| Github.workflow Test contexts
| Reusable Github.job reusable-test
| Reusable Github.workflow Reusable test contexts
```

--> github.workflow does not help here because we still don't get the name of the calling workflow.

fmauch, 2024-04-16:
I'll try to have a look tonight

PR #49 (2024-04-16): ICI: Remove caching of workspace
Fixes #33 the easiest way: just don't cache the workspace, but only ccache directory.

> Does it make sense to cache the workspace at all, how much is the speed benefit compared to only caching ccache?
> 
> * Locally, it takes 20s vs 2min if I remove the build folder.
> * [cache](https://github.com/ros-controls/ros2_control/actions/runs/8707803628/job/23883672248?pr=1485) vs [no cache](https://github.com/ros-controls/ros2_control/actions/runs/8707803627/job/23884339041?pr=1485) of ros2_control binary build: The ICI step takes the same time.

This also resolved  [... cut]
closed 2024-04-16 by christophfroehlich