ros-controls/ros2_control_ci #34: ICI workflow @ humble is flaky

christophfroehlich, 2024-03-16:
Why does it fail here? 👀 @saikishor 
https://github.com/ros-controls/ros2_control_ci/actions/runs/8304468913/job/22730143420

    [ RUN      ] TestLoadController.multi_ctrls_test_type_in_param
    ...
    [INFO] [1710554585.628199404] [test_controller_manager]: Configuring controller 'ctrl_1'
    terminate called after throwing an instance of 'std::runtime_error'
      what():  Can not get command interface configuration until the controller is configured.

And why was no issue reported @fmauch? 🤔

christophfroehlich, 2024-03-20:
@fmauch can this be the result of https://github.com/ros-controls/ros2_control_ci/issues/33? (code from master from the compatibiltiy-workflow sneaked into humble workspace)

christophfroehlich, 2024-04-19:
Caching is now fixed (I hope) but the issue persists https://github.com/ros-controls/ros2_control_ci/actions/runs/8747264205/job/24005451302

christophfroehlich, 2024-04-28:
See #62

christophfroehlich, 2024-04-29:
and see #63 (happens on main as well as on testing)

saikishor, 2024-04-29:
Once the jazzy stuff is done, I hope I'll have time to take a look at this

christophfroehlich, 2024-07-18:
Reported in https://github.com/ros-controls/ros2_control/issues/1368

closed 2024-07-18 by christophfroehlich