ros-controls/ros2_control_ci #37: ICI_workflow: No issues are reported any more

christophfroehlich, 2024-03-17:
Humble workflow fails during testing, like here
https://github.com/ros-controls/ros2_control_ci/actions/runs/8304468913/job/22730143420

[ RUN      ] TestLoadController.multi_ctrls_test_type_in_param
...
[INFO] [1710554585.628199404] [test_controller_manager]: Configuring controller 'ctrl_1'
terminate called after throwing an instance of 'std::runtime_error'
  what():  Can not get command interface configuration until the controller is configured.

Why are no issues reported any more? 🤔

saikishor, 2024-03-17:
@christophfroehlich this is weird there is no logic changed in the code. It Is very unlikely that this happens. I'll take a look and get back to you

fmauch, 2024-03-17:
I guess it relates to the condition to be precise:

https://github.com/ros-controls/ros2_control_ci/blob/370595a82edd94e6ad73fabf7855d23f8079be86/.github/workflows/reusable-industrial-ci-with-cache.yml#L119

That only triggers on a build test or on a failed test. However, in the run test-results seem not to be evaluated due to the exception being raised. Probably we should add checking for `run_target_test` having a return code `!=0`

fmauch, 2024-03-17:
I tried to reproduce this locally, but I didn't succeed. Adding a segfault to a test or raising an exception all resulted in `colcon test` step returning `0` and `colcon test-result` returning `1`. I also tried generating a timeout during a test, same result. `colcon test-result` returning `1` should result in `steps.ici.outputs.target_test_results == '1'` and therefore create an issue.

fmauch, 2024-03-18:
https://github.com/ros-industrial/industrial_ci/issues/853

christophfroehlich, 2024-04-28:
Seems to be fixed with https://github.com/ros-industrial/industrial_ci/pull/854 :+1:

closed 2024-04-28 by christophfroehlich