ros-controls/ros2_control_ci #43: Auto-issue for failed pipelines reaches API limit

christophfroehlich, 2024-04-04:
https://github.com/ros-controls/ros2_control_ci/actions/runs/8548124997/job/23421377322

> You have exceeded a secondary rate limit. Please wait a few minutes before you try again. 

I could imagine that this happens when several jobs of a matrix-strategy finish at the same time and the action-bot interactions have a rate limit.

fmauch, 2024-04-07:
I found this issue upstream: https://github.com/JasonEtco/create-an-issue/issues/142

[This comment](https://github.com/JasonEtco/create-an-issue/issues/142#issuecomment-1506159876) suggests using another type of token to improve things. Seems like currently no clear solution exists.

christophfroehlich, 2024-04-07:
Could we find a workaround with some random time delay before trying to create an issue?